BBN-Q/PyQLab #20: RabiWidth giving a fatal cache controller error

calebhowington, 2015-01-25:
Compiling a pulse with

    RabiWidth(q1, 1e-9*np.linspace(0, 1e3, 101), showPlot=True)

Leads to a red L2 on the APS producing the measurement pulse.

matthewware, 2015-01-26:
What happens if you power cycle?  Are you still using a 5 us pulse and 19 MHz SSB?

caryan, 2015-01-26:
We're having trouble reproducing this issue.  What firmware version are these APS2's running?

matthewware, 2015-01-26:
If these are ones we programmed successfully the first time they should be flashed with v2.24 that came with the zipped v0.5 release.  The one we had to redo is running a slightly newer version... v2.27 I think.  My guess is 2.24.  Is that right Caleb?

calebhowington, 2015-01-26:
Power cycle revives the unit, problem happens again when we try to run the experiment. L1 flashes as normal, and then we get a solid green L1 and solid red L2. Still using a 5us pulse with a 19MHz SSB.

This one is running the v2.24 that Matt and I flashed, we haven't used the v2.27 one yet.

caryan, 2015-01-26:
v2.27 has a bug so stick with v2.24 for now.  I think you might be exceeding the waveform cache (128k samples) because the 19MHz autodyne frequency is not a nice multiple of 10ns.  As an aside for measurement channels, the physical channel SSB frequency should be zero and the frequency should be set on the logical measurement channel.

Could you try a couple different approaches:
* reduce the measurement pulse width to 1us 
* change the autodyne frequency to a multiple of 10MHz to be a nice multiple of the 10ns step.

mattai1986, 2015-01-26:
Hi, Thanks for this suggestion. The RabiWidth sequence runs fine with a 10 MHz autodyne freq. However, we still have to set this freq under the SSB freq in the Physical tab of Channels. Setting the autodyne freq to 10 MHz in the logical tab and the SSB freq to zero leads to a baseband measurement pulse when I compile the sequence. I tried restarting the GUI as well as iphython and that had no effect.

mattai1986, 2015-01-26:
Ok, so changing autodyne freq does work now. Not sure what I did wrong the first few times. 

Thanks.

matthewware, 2015-01-26:
Do you remember what you did?  I've seen this issue before where the autodyne frequency doesn't seem to have an effect.

caryan, 2015-01-26:
Autodyne frequency is one of these ones that doesn't automagically update from the GUI to the notebook. There is work in progress to fix this.

caryan, 2015-01-26:
This will be fixed with the PREFETCH command but for now sequences must fit inside a single waveform cache of 128k samples.  Error added in 3e9c74d07793c93d5df8788e3064f6ce68bec897

closed 2015-01-26 by caryan