BBN-Q/PyQLab #26: Anaconda 2.2.0 h5py import error

bcdonovan, 2015-04-01:
While attempting to follow the example notebook I received the following error from the startup.py script:

```
Traceback (most recent call last):
  File "startup.py", line 2, in <module>
    from Libraries import instrumentLib, channelLib
  File "C:\Projects\Q\src\PyQLab\Libraries.py", line 7, in <module>
    from QGL.Channels import ChannelLibrary
  File "C:\Projects\Q\src\PyQLab\QGL\__init__.py", line 1, in <module>
    from Channels import QubitFactory, MeasFactory, Qubit
  File "C:\Projects\Q\src\PyQLab\QGL\Channels.py", line 25, in <module>
    import Compiler
  File "C:\Projects\Q\src\PyQLab\QGL\Compiler.py", line 35, in <module>
    from APSPattern import write_APS_file
  File "C:\Projects\Q\src\PyQLab\QGL\APSPattern.py", line 19, in <module>
    import h5py
  File "C:\Apps\Anaconda\lib\site-packages\h5py\__init__.py", line 13, in <module>
    from h5py import _conv
  File "h5py\h5r.pxd", line 21, in init h5py._conv (D:\Build\h5py\h5py-2.4.x\h5py\_conv.c:6809)
  File "h5py\_objects.pxd", line 12, in init h5py.h5r (D:\Build\h5py\h5py-2.4.x\h5py\h5r.c:2824)
  File "h5py\_objects.pyx", line 1, in init h5py._objects (D:\Build\h5py\h5py-2.4.x\h5py\_objects.c:6927)
ImportError: DLL load failed: The specified procedure could not be found.
```

Adding 
```
import h5py
```
as the first line of the startup.py script eliminates the error.

bcdonovan, 2015-04-01:
A bit more detail:

from a python shell

```
import h5py
```

works

```
import QGL.APSPattern
```

results in the error.

bcdonovan, 2015-04-01:
More specifically from the python shell:

```
import h5py
```

works.

``` 
import QGL.APSPattern
```

fails but the active line of APS2Pattern is the import of h5py

caryan, 2015-04-02:
Also present on Linux updating to Anaconda 2.2.  Although there even more broken:

```python
Python 2.7.9 |Anaconda 2.2.0 (64-bit)| (default, Mar  9 2015, 16:20:48) 
Type "copyright", "credits" or "license" for more information.

IPython 3.0.0 -- An enhanced Interactive Python.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://binstar.org
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import h5py
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-e81388eeaef8> in <module>()
----> 1 import h5py

/home/cryan/anaconda/lib/python2.7/site-packages/h5py/__init__.py in <module>()
      8 #           and contributor agreement.
      9 
---> 10 from h5py import _errors
     11 _errors.silence_errors()
     12 

ImportError: libhdf5.so.8: cannot open shared object file: No such file or directory
```

h5py seems to be built against libhdf5.so.8 but .9 is installed in anaconda/lib

caryan, 2015-04-02:
Hmmm... Must have been something goofed in my install.  A clean 2.2 install on Linux doesn't display this bug.

caryan, 2015-04-02:
It's definitely present in a clean install of Anaconda 2.2. It's a conflict between pytables and hdf5. Since pytables is loaded by bokeh the import order matters. @bcdonovan's fix will work for now but pytables will crash if it is ever used.

I've filed an [issue](https://github.com/ContinuumIO/anaconda-issues/issues/300) with Anaconda.

bcdonovan, 2015-04-02:
This impacts ExpSettingsGUI.py as well. I needed to add the import h5py as the very first line.

caryan, 2015-07-06:
I think this is fixed as of Anaconda 2.3.  We'll reopen if it crops up again.

closed 2015-07-06 by caryan