BBN-Q/PyQLab #31: Automate unit test

bcdonovan, 2015-04-28:
Expand upon current unit tests.
Goal is to be able to apply continuous integration to PyQlab
Factor out show from existing unit tests so that they may be automated

caryan, 2015-04-28:
That would be pretty slick with a Travis badge.

bcdonovan, 2015-04-28:
That's the goal

caryan, 2015-04-28:
What I've done in other pulse sequence languages is have a h5 file with reference compiled sequences that is loaded by the test fixture to check against.  Then there is an interactive/non-interactive testing mode which plots or doesn't plot the sequences.

bcdonovan, 2015-04-28:
Someone with admin rights will need to enable the travis-ci.org to github connection. I do not have the rights to enable PyQLab in travis.

caryan, 2015-04-28:
Done

matthewware, 2015-06-02:
This looks nice Brian.  Should we close this issue?  Seems like Travis is working well with repo.

bcdonovan, 2015-06-03:
Lets keep this open for now. I would like to add a few more unit tests as well as having travis ci working.

blakejohnson, 2015-06-04:
Right, the test_QGL pieces still need to be converted to waveform masks.

bcdonovan, 2015-06-05:
@blakejohnson Do you think the new test_QGL implementation is sufficient to close this or do you have more test work in mind?

blakejohnson, 2015-06-05:
In terms of automation, this is sufficient. We should probably next look at code coverage (i.e. get coveralls.io up and running).

blakejohnson, 2015-07-10:
This has been really great for bug fixing in the last week. Thanks, Brian.

closed 2015-07-10 by blakejohnson

bcdonovan, 2015-07-10:
Happy to Help