BBN-Q/PyQLab #33: Passing validation when starting from scratch

blakejohnson, 2015-04-29:
Post merge of the explinter branch, I cannot seem to pass validation when starting from scratch with no instruments and no channels. The issue seems to be related to the physical channel -> AWG mapping, because after creating an AWG (e.g. and APS1) and then populating some physical channels, when I choose the APS1 as the AWG and then highlight another channel and return, the choice of AWG seems to vanish.

caryan, 2015-04-29:
Okay, tracked it down.  It's a the classic reference binding vs copy constructor we've run into many times before.  If you add these lines to ``ChannelView.enaml`` [here]https://github.com/BBN-Q/PyQLab/blob/a72845e8266032a31c8b53cf24c843d8a045d391/QGL/ChannelsViews.enaml#L92)

```python
            index ::
                setIfNotEmpty(chan.AWG, instrumentLib, selected_item)
                print("id of chan.AWG {}".format(id(chan.AWG)))
                print("id of instrument {}".format(id(instrumentLib[selected_item])))
```
you'll see they have different pointers.  We solved this before by passing higher level objects and just updating the member. 

@bcdonovan this is probably from ``setIfNotEmpty``.  Maybe you can see a clever way to keep the repeated code smell away and not run into this issue.

blakejohnson, 2015-04-29:
Fixed by #35.

closed 2015-04-29 by blakejohnson