BBN-Q/PyQLab #38: Ipython notebook kernel restart required after working in ExpSettingsGUI?

bhassick, 2015-05-08:
This may be an issue specifically with notebook, however, when working with ipyhton notebooks, modifying settings via the ExpSettingsGUI and writing them out does not result in them being read in the next time run startup is invoked on an open notebook.  The only way to read in the new changes is to restart the kernel in the notebook with which you are interacting.  

This may be known or expected behavior, but I thought it was of interest to report here for the sake of lessons learned.

caryan, 2015-05-11:
This could probably be fixed by if blocking when we bind the libraries [here](https://github.com/BBN-Q/PyQLab/blob/ea4f6913d603b4c02c46d619ae115f9a6c1d4ff8/startup.py#L6).  It's probably leaving ``watchdog`` updating a stale reference.

caryan, 2015-05-12:
@bhassick Could you give steps that give this issue.  I couldn't reproduce it on my machine.

blakejohnson, 2015-05-13:
I also cannot reproduce on my machine.

bcdonovan, 2015-05-14:
I was able to reproduce on a Windows 7 64-bit machine.
 (Anaconda 2.2.0, python 2.7.9,  ipython 3.0.0-f75fda4)

I ran the following in the ipython notebook
```
cd PYQLAB_HOME
run startup.py
import ExpSettingsVal
ExpSettingsVal.list_config()
```

This will print the current config to the notebook.

Edited config in ExpSettingsGUI and ran

```
run startup.py
ExpSettingsVal.list_config()
```

Changes were not visible until I restarted the ipython kernel.

bcdonovan, 2015-05-14:
This may be related to the code [here](https://github.com/BBN-Q/PyQLab/blob/f64af23fd695bb6b93773848a4d065de311242d2/instruments/InstrumentManager.py#L94-100).

This code appears to update entries in the library only if they exist (by name). I don't think adding, removing, or deleting items in the config file will be reflected in the underlying dictionary. 

I will add that I was able to update an existing instrument without restarting ipython.

caryan, 2015-05-14:
Yes, that's true that only existing instrument/channels are updated. I guess that's all I was checking.

commit 34da0d20 (2015-05-15): Merge pull request #40 from BBN-Q/hotfix/filewatch

Looks good. Closes #38.
closed 2015-05-15 by caryan