BBN-Q/PyQLab #39: In-line notebook plotting is broken in iPython 3.0/ Jupyter

caryan, 2015-05-12:
Our notebook checking function [in_pynb()](https://github.com/BBN-Q/PyQLab/blob/d4007c4745db79971da01b02818fcdff2fc10588/QGL/Plotting.py#L26) doesn't work anymore.

caryan, 2015-05-12:
In particular, [this iPython issue](https://github.com/ipython/ipython/pull/4980) killed the hook we were using.  They really seem to want to force you to use display system and that's it.

blakejohnson, 2015-05-13:
I guess we can make `startup()` take an argument about whether to enable inline plotting. Maybe `startup(inline=False)` to make the default thing still work at the console?

caryan, 2015-05-13:
I think I have another hack workaround by checking whether instantiating a widget throws but I need to double check it doesn't break iPython 2.x

caryan, 2015-05-20:
Hack workaround doesn't work in iPython 2.x so I think we'll have to adopt the ``output_notebook``, ``output_file`` convention and have the user manually start interactive plotting.

blakejohnson, 2015-05-21:
That's fine with me.

matthewware, 2015-05-27:
So should we have people run ``output_notebook`` before trying to plot anything in IPython 3?

caryan, 2015-05-27:
No, I haven't committed that fix yet.  It's currently just broken.

blakejohnson, 2015-05-29:
Close?

caryan, 2015-05-29:
Yep.  I didn't get the magic words right in the commit message for 09e5fd8f79b06e19419fae3773ea12e7ede7d2ae.  I used "Closes issue" instead of just "closes.

closed 2015-05-29 by caryan