BBN-Q/PyQLab #50: UnicodeDecodeError

dieris, 2015-06-17:
Error when trying to print a sequence in an IPython notebook (before compiling to hardware).

Using Python 2.7 in windows 7 64 bit. 

Example of error:

C:\Users\qlab\Anaconda\lib\json\encoder.pyc in encode(self, o)
    208         if not isinstance(chunks, (list, tuple)):
    209             chunks = list(chunks)
--> 210         return ''.join(chunks)
    211 
    212     def iterencode(self, o, _one_shot=False):

UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 9: ordinal not in range(128)

bcdonovan, 2015-06-17:
Diego, can you zip up your json files and your notebook and e-mail them to me to take a look at?

dieris, 2015-06-17:
Colm and I found that the error only appears in IPython 3.0. In 3.1 there is no error, but the tensor operator * [example: seq = Id(q1)*Id(q2)] is displayed as "âŠ—"

caryan, 2015-06-18:
In particular this [iPython PR](https://github.com/ipython/ipython/pull/7853) "fixed" it. Like @dieris noted it's bizarre that the ⊗ doesn't appear correctly anymore but since it's readable I'm closing the issue for now.

closed 2015-06-18 by caryan