BBN-Q/PyQLab #58: Sequences with only marker data do not wait for trigger on APS1

caryan, 2015-07-09:
Reported by @dieris

blakejohnson, 2015-07-09:
I have a strong suspicion that this is related to the new test failure we see with de5d147a20ea232c51a99edd35a1c0f140caad00. Somehow we are creating markers now that align or overhang the last IQ waveform point.

blakejohnson, 2015-07-10:
So, the new test failure was probably a red herring. @dieris can you post your channel parameters for the digitizerTrig and slaveTrig channels (both logical and physical)?

dieris, 2015-07-10:
These are the relevant channel settings, I believe. 
BBNAPS2 is the master (with markers only)
BBNAPS4 is the slave, with pulses for q3 and M-q3

"slaveTrig": {
      "enabled": true, 
      "label": "slaveTrig", 
      "physChan": "BBNAPS2-1m1", 
      "pulseParams": {
        "length": 1e-08, 
        "shapeFun": "square"
      }, 
      "x__class__": "LogicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }
  }, 
"digitizerTrig": {
      "enabled": true, 
      "label": "digitizerTrig", 
      "physChan": "BBNAPS2-4m1", 
      "pulseParams": {
        "length": 8.33e-10, 
        "shapeFun": "square"
      }, 
      "x__class__": "LogicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 

  "BBNAPS2-1m1": {
      "AWG": "BBNAPS2", 
      "delay": -5e-08, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS2-1m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
 "BBNAPS2-4m1": {
      "AWG": "BBNAPS2", 
      "delay": 0.0, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS2-4m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
"BBNAPS4-12": {
      "AWG": "BBNAPS4", 
      "IChannel": "", 
      "QChannel": "", 
      "SSBFreq": -100000000.0, 
      "ampFactor": 0.8405, 
      "delay": 0.0, 
      "enabled": true, 
      "generator": "Q3gate", 
      "label": "BBNAPS4-12", 
      "phaseSkew": 7.51, 
      "x__class__": "PhysicalQuadratureChannel", 
      "x__module__": "QGL.Channels"
    }, 
    "BBNAPS4-1m1": {
      "AWG": "BBNAPS4", 
      "delay": 0.0, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS4-1m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
    "BBNAPS4-2m1": {
      "AWG": "BBNAPS4", 
      "delay": 0.0, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS4-2m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
    "BBNAPS4-34": {
      "AWG": "BBNAPS4", 
      "IChannel": "", 
      "QChannel": "", 
      "SSBFreq": 0.0, 
      "ampFactor": 0.9601, 
      "delay": 0.0, 
      "enabled": true, 
      "generator": "Autodyne_Mq3", 
      "label": "BBNAPS4-34", 
      "phaseSkew": -0.61, 
      "x__class__": "PhysicalQuadratureChannel", 
      "x__module__": "QGL.Channels"
    }, 
    "BBNAPS4-3m1": {
      "AWG": "BBNAPS4", 
      "delay": 0.0, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS4-3m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
    "BBNAPS4-4m1": {
      "AWG": "BBNAPS4", 
      "delay": 0.0, 
      "enabled": true, 
      "gateBuffer": 0.0, 
      "gateMinWidth": 0.0, 
      "label": "BBNAPS4-4m1", 
      "x__class__": "PhysicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
"M-q3": {
      "autodyneFreq": 9000000.0, 
      "enabled": true, 
      "gateChan": "M-q3-gate", 
      "label": "M-q3", 
      "measType": "autodyne", 
      "physChan": "BBNAPS4-34", 
      "pulseParams": {
        "amp": 0.9, 
        "buffer": 4e-08, 
        "cutoff": 2, 
        "length": 2.5e-06, 
        "shapeFun": "tanh", 
        "sigma": 2e-09
      }, 
      "x__class__": "Measurement", 
      "x__module__": "QGL.Channels"
    }, 
    "M-q3-gate": {
      "enabled": true, 
      "label": "M-q3-gate", 
      "physChan": "BBNAPS4-2m1", 
      "pulseParams": {
        "length": 1e-07, 
        "shapeFun": "square"
      }, 
      "x__class__": "LogicalMarkerChannel", 
      "x__module__": "QGL.Channels"
    }, 
 "q3": {
      "enabled": true, 
      "label": "q3", 
      "physChan": "BBNAPS4-12", 
      "pulseParams": {
        "buffer": 0.0, 
        "cutoff": 2, 
        "dragScaling": -0.14, 
        "length": 6e-08, 
        "pi2Amp": 0.4064, 
        "piAmp": 0.8089, 
        "shapeFun": "drag", 
        "sigma": 5e-09
      }, 
      "x__class__": "Qubit", 
      "x__module__": "QGL.Channels"
    }

commit c725cf1c (2015-07-13): Ensure we WAIT when merging markers onto empty sequences.

Fixes #58.

--BRJ and CAR
closed 2015-07-13 by CoherenceController