BBN-Q/PyQLab #79: RFC: bring test_data into the repo

caryan, 2015-09-25:
Currently updating the unit testing golden sequences is messy and it's a few extra hidden steps for users to get the test_data to run on any changes they make before pushing. The compressed test data is only 88kB (and only 51kB if we switch to bzip2) and I'd be okay to bring it into the repo for now. 

@bcdonovan and @blakejohnson any thoughts?

matthewware, 2015-09-25:
I think it sounds like a good idea for now.  I'd like to be able to test locally before pushing to github and anxiously refreshing while Travis works.  I might also suggest that we move the config json files or at least the paths to the test folder.  Currently running the tests overwrites/clears your local config files.  This makes sense for Travis, but probably not for local testing.

caryan, 2015-09-25:
It looks like GitHub will give BBN-Q 1GB for git large file storage [link](https://github.com/blog/1986-announcing-git-large-file-storage-lfs).  I've requested access for BBN-Q and it seems like the best long-term solution if the test data size ever grows.  Github bundles ``git-lfs`` into the Github for Windows so it should be easy for users.

caryan, 2015-09-25:
@matthewware can you file a separate issue for the config files and ``test_json`` issue.

bcdonovan, 2015-09-25:
I'm comfortable bringing the test data into the repo. I like the Github LFS option but we might need to watch the bandwidth quota rather than the file size limit. I think we are currently setup for Travis will run for every push back to github. We are probably ok for now given the file size but it may start to add up.

matthewware, 2015-10-01:
Looks like git-lfs is available today for all repos: https://github.com/blog/2069-git-large-file-storage-v1-0
Bandwidth is capped at 1GB per month: https://help.github.com/articles/billing-plans-for-git-large-file-storage/

matthewware, 2015-10-01:
I've also been wondering if we could just create hashes for the h5 files and then check the newly created files against them?

blakejohnson, 2015-10-01:
@matthewware that would be sufficient for a simple pass/fail test. But when you fail, it is nice to have some discoverability of *where* you are failing. That's why Brian went with storing the whole file.

caryan, 2015-10-20:
There's some work a  ``feature/git-lfs-test-files`` branch (5a60a111854e72706a72807d160059032aea8d20) with this at least working with git-lfs.  I've spent a couple hours trying to get git-lsf to work with Travis and after some discussion with @blakejohnson I think we'll just leave it until the support from Travis is better.  For future reference some notes:

* there seem to be only two [other](https://gist.github.com/lbergelson/4265e5ef3ec8b60dac97) [projects](https://github.com/piwik/piwik/pull/8392) trying to do this which isn't a good sign
* the very first thing travis does is checkout the project code so if git-lfs isn't installed from the start you'll have to install it and go back and ``git lfs fetch``
* the travis CI guys made a stab at including git-lfs by default [here](https://github.com/travis-ci/travis-ci/issues/3634) however this was only version 0.5 which has [token authentication problems](https://github.com/github/git-lfs/issues/598)
* to top it off the travis build environment update from April 2015 that included this [was cancelled](http://docs.travis-ci.com/user/build-environment-updates/2015-04-09/).  There have been no updates since despite one being promised in June
* the only option then is to install git-lfs but this requires sudo which means no fast container runs

caryan, 2016-01-05:
An update from looking into this again.  Travis have said nothing about any more updates to the 12.04 images.  They have however released a 14.04 image using GCE that comes with sudo [blog post](https://blog.travis-ci.com/2015-10-14-opening-up-ubuntu-trusty-beta) .

blakejohnson, 2016-01-05:
And does this new image have git lfs?

blakejohnson, 2016-04-07:
It looks like it should now be possible to do install Git lfs after repo checkout and then to `git lfs pull`. See:
https://github.com/github/git-lfs/issues/325

caryan, 2016-04-07:
Sure, that's what we were doing [here](https://github.com/BBN-Q/PyQLab/commit/5a60a111854e72706a72807d160059032aea8d20) but the question was how do you install git lfs without sudo?

blakejohnson, 2016-04-07:
Can't you install to `/usr/local` without sudo? I thought that was just needed to upgrade git prior to cloning.

caryan, 2016-04-26:
Closed over in [QGL #25](https://github.com/BBN-Q/QGL/pull/25)

closed 2016-04-26 by caryan